adeept/adeept_picarpro #4: unable to start webServer after fresh installation

AdrianPeniak, 2022-03-23:
Dear everyone,

I prepare & install sd card according instructions and everything pass. However, after reboot webServer does not start. When I start it manually a get following error:  

`pi@raspberrypi:~ $ sudo python3 //home/pi/adeept_picarpro/server/webServer.py
......................pause..........................
mpu6050 disconnected, ARM MODE ON
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "//home/pi/adeept_picarpro/server/webServer.py", line 26, in <module>
    import app
  File "/home/pi/adeept_picarpro/server/app.py", line 8, in <module>
    from camera_opencv import Camera
  File "/home/pi/adeept_picarpro/server/camera_opencv.py", line 2, in <module>
    import cv2
  File "/usr/local/lib/python3.9/dist-packages/cv2/__init__.py", line 5, in <module>
    from .cv2 import *
ImportError: numpy.core.multiarray failed to import`

Thanks for any advice.

adeept, 2022-03-24:
Hello,
It may be an incompatibility problem with some dependent libraries.

Please try to burn the official old version of the Raspberry Pi image file (2021-05-07), the image download link (ZIP file):
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-05-28/

Or burn the image file we provide. The dependent library has been installed in the image file and can be used after configuring WiFi and SSH. There is no need to run the setup.py program.
The image files are in the "PiCar-Pro Image" PDF linked below.
Link: https://www.adeept.com/learn/detail-50.html

If the error message "ws2811_init failed with code -3 (Hardware revision is not supported)" appears when running
Please enter in the Raspberry Pi command line:
sudo pip3 install rpi_ws281x
sudo pip3 install rpi-ws281x --upgrade

Best!

Technical Support
***@***.***

Thanks for any advice.

AdrianPeniak, 2022-03-24:
Thanks @adeept for your fast response. I fixed it on newest image (debian 11) with:
_sudo pip3 install opencv-contrib-python==3.4.3.18
sudo pip3 install opencv-python==3.4.11.45_ 

now everything works fine. Thanks a lot

closed 2022-03-24 by AdrianPeniak