adeept/adeept_picarpro #6: Could not determine default I2C bus for platform

Nick-Harvey, 2022-04-23:
Hi there, 

I got the Picar Pro and I'm trying to get the webServer running and I get the following error: 

```
Traceback (most recent call last):
  File "/home/nick/git/adeept_picarpro/server/webServer.py", line 14, in <module>
    import RPIservo
  File "/home/nick/git/adeept_picarpro/server/RPIservo.py", line 17, in <module>
    pwm = Adafruit_PCA9685.PCA9685()
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_PCA9685/PCA9685.py", line 74, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
```

**Additional Setup information**

Python version: Python 3.9.2
Package list:
```
Package                 Version
----------------------- -----------
Adafruit-Blinka         7.2.1
Adafruit-GPIO           1.0.3
Adafruit-PCA9685        1.0.1
Adafruit-PlatformDetect 3.22.1
Adafruit-PureIO         1.1.9
adafruit-python-shell   1.3.3
arandr                  0.1.10
args                    0.1.0
astroid                 2.5.1
asttokens               2.0.4
automationhat           0.2.0
beautifulsoup4          4.9.3
blinker                 1.4
blinkt                  0.1.2
buttonshim              0.0.2
Cap1xxx                 0.1.3
cbor2                   5.4.2.post1
certifi                 2020.6.20
chardet                 4.0.0
click                   7.1.2
clint                   0.5.1
colorama                0.4.4
colorzero               1.1
cryptography            3.3.2
cupshelpers             1.0
dbus-python             1.2.16
Deprecated              1.2.13
distro                  1.5.0
docutils                0.16
drumhat                 0.1.0
envirophat              1.0.0
ExplorerHAT             0.4.2
Flask                   1.1.2
Flask-Cors              3.0.10
fourletterphat          0.1.0
gpiozero                1.6.2
html5lib                1.1
idna                    2.10
imutils                 0.5.4
isort                   5.6.4
itsdangerous            1.1.0
jedi                    0.18.0
Jinja2                  2.11.3
lazy-object-proxy       0.0.0
logilab-common          1.8.1
luma.core               2.3.1
luma.oled               3.8.1
lxml                    4.6.3
MarkupSafe              1.1.1
mccabe                  0.6.1
microdotphat            0.2.1
mote                    0.0.4
motephat                0.0.3
mpu6050-raspberrypi     1.2
mypy                    0.812
mypy-extensions         0.4.3
numpy                   1.19.5
oauthlib                3.1.0
olefile                 0.46
pantilthat              0.0.7
parso                   0.8.1
pexpect                 4.8.0
pgzero                  1.2
phatbeat                0.1.1
pianohat                0.1.0
picamera                1.13
piglow                  1.2.5
pigpio                  1.78
Pillow                  8.1.2
pip                     22.0.4
psutil                  5.8.0
pybase64                1.2.1
pycairo                 1.16.2
pycups                  2.0.1
pyftdi                  0.54.0
pygame                  1.9.6
Pygments                2.7.1
PyGObject               3.38.0
pyinotify               0.9.6
PyJWT                   1.7.1
pylint                  2.7.2
pyOpenSSL               20.0.1
pyserial                3.5b0
pysmbc                  1.0.23
python-apt              2.2.1
pyusb                   1.2.1
pyzmq                   22.3.0
rainbowhat              0.1.0
reportlab               3.5.59
requests                2.25.1
requests-oauthlib       1.0.0
responses               0.12.1
roman                   2.0.0
RPi.GPIO                0.7.0
rpi-ws281x              4.3.4
RTIMULib                7.2.1
scrollphat              0.0.7
scrollphathd            1.2.1
Send2Trash              1.6.0b1
sense-hat               2.2.0
setuptools              62.1.0
simplejson              3.17.2
six                     1.16.0
skywriter               0.0.7
smbus2                  0.4.1
sn3218                  1.2.7
soupsieve               2.2.1
spidev                  3.5
ssh-import-id           5.10
sysv-ipc                1.1.0
thonny                  3.3.14
toml                    0.10.1
touchphat               0.0.1
twython                 3.8.2
typed-ast               1.4.2
typing-extensions       3.7.4.3
unicornhathd            0.0.4
urllib3                 1.26.5
webencodings            0.5.1
websockets              10.3
Werkzeug                1.0.1
wheel                   0.34.2
wrapt                   1.12.1
zmq                     0.0.0
```

A quick google search says that this is using a deprecated Adafruit_Python_GPIO library and this error keeps you from getting past the first few steps of the build.

adeept, 2022-04-25:
Hello,
It may be a problem with the version of the Raspberry Pi image file.
Please try the following two methods:
1. Download Raspberry Pi OS (Legacy):
https://www.raspberrypi.com/software/operating-systems/
Raspberry Pi official description of the Raspberry Pi OS (Legacy):
https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

2. Download the previous Raspberry Pi version, Raspberry Pi image file (2021-5-07) download address:
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-05-28/

Best!

Technical Support
***@***.***

Nekkit, 2022-09-04:
I have the same problem. Picar pro bought in September 22, with a RPI 4b. 

Installed the official RPI OS tried the legacy and non legacy and didn't solve the problem

Nekkit, 2022-09-04:
Same issue here
RPI 4b 
Official legacy OS and normal OS both same issue.
Did you manage to fix this?

adeept, 2022-09-05:
Hello,
This is due to the incompatibility between the Raspberry Pi image file and some dependent libraries.
Please download and burn the Raspberry Pi OS (Legacy) according to the content "Method one" in the link below, then configure SSH and WiFi, and run the setup.py program after downloading the product code.
Link: https://www.adeept.com/learn/tutorial-247.html

Best!

Technical Support
***@***.***

---- Replied Message ----
From ***@***.***>Date 9/4/2022 10:00To ***@***.***>Cc ***@***.***> ,
 ***@***.***>Subject Re: [adeept/adeept_picarpro] Could not determine default I2C bus for platform (Issue #6)

I have the same problem. Picar pro bought in September 22, with a RPI 4b.
Installed the official RPI OS tried the legacy and non legacy and didn't solve the problem