adeept/adeept_picarpro #7: Unsupport opencv-contrib-python==3.4.3.18 library

JPHuls, 2022-05-03:
Please update the setup.py to use a more recent version.

adeept, 2022-05-05:
Hello,
Please make sure you are using the correct version of the Raspberry Pi image.
Please use Raspberry Pi OS (Legacy).
Raspberry Pi official description of the Raspberry Pi OS (Legacy):
https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

visit the official website of the Raspberry Pi through a browser to download Raspberry Pi OS (Legacy): https://www.raspberrypi.com/software/operating-systems/

Best!

Technical Support
***@***.***

JPHuls, 2022-05-05:
Thank you for your feedback. I was using Raspberry Pi OS Desktop as mention in the instruction. However, there were issues with the opencv library:
"ERROR: Could not find a version that satisfies the requirement opencv-contrib-python==3.4.3.18 (from versions: 3.4.11.45, 3.4.13.47, 3.4.14.51, 3.4.15.55, 3.4.16.59, 3.4.17.61, 3.4.17.63, 4.4.0.46, 4.5.1.48, 4.5.2.52, 4.5.3.56, 4.5.4.58, 4.5.4.60, 4.5.5.62, 4.5.5.64)
ERROR: No matching distribution found for opencv-contrib-python==3.4.3.18"
Meaning that the used opencv library is no longer available at the locations used in the setup script. Trying to load the next available opencv version seemed to work, but then WebServer.py failed on one of the opencv methods.

Something more people seem to have issues with according to posts I found. In the end I had to revert to the (outdated - 092020) picarpro image (adeeptPiCarPro.img) which you provided in one of the links. Which works btw. But still, after all the time I spent, it's a bit of a discouragement.

NB: I'm myself an IT professional with some software development experience... so I'm wondering how this woman and her little daughter, as used in one of the product photos, handled this issue 😉

So, please and if not to much of a hassle, update both the installation scripts and python code to support a more recent opencv version.

Kind regards,

Jean-Pierre Huls

https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

adeept, 2022-10-11:
Hello,
thanks for your suggestion.
Since the python version in different Raspberry Pi image files will change, different versions of python will lead to changes in the versions of dependent libraries such as "numpy" and "opencv" that are compatible with it, and even problems such as finding a specific version of the dependent library may occur.
(The previous Raspberry Pi image used python3.7 version, and the recent Raspberry Pi image used python3.9 version. With the update of the image file, subsequent Raspberry Pi images will also use the new python version. Therefore, there will be incompatibility problems with dependent libraries).

Raspberry Pi has officially launched a Raspberry Pi image that will not change the python version (python3.7)---Raspberry Pi OS (Legacy).
Using this Raspberry Pi image there is no/few incompatibilities with dependencies.

Best!

Technical Support
***@***.***

 https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/