adeept/adeept_picarpro #8: webserver.py issue

JPHuls, 2022-05-29:
Hi,

All the individual test code work like a charm. However, with the webserver I seem to experience an issue. When opening the browser, the dashboard does show, including the video feed. However, after a few seconds at the bottom of the "Adeept Bot Contorller" (where can I correct this type??) a popup appears saying "Connect failed - Reconnecting"; the reconnecting part nevers seems to succeed.

What am I missing.

I'm using a Pi 3 model B+ and your image with all the software preïnstalled.

Kind regards,

Jean-Pierre

JPHuls, 2022-05-30:
Is my assessment and solution as described in UPDATE part correct?

adeept, 2022-05-31:
Hello,
This is due to insufficient power supply causing the Raspberry Pi to reboot.
Please use 18650 battery for power supply. If an 18650 battery is already in use, use the 18650 battery charger to fully charge the battery.

Best!

Technical Support
***@***.***

---- Replied Message ----
From ***@***.***>Date 5/30/2022 00:05To ***@***.***>Cc ***@***.***>Subject [adeept/adeept_picarpro] webserver.py issue (Issue #8)

Hi,
All the individual test code work like a charm. However, with the webserver I seem to experience an issue. When opening the browser, the dashboard does show, including the video feed. However, after a few seconds at the bottom of the "Adeept Bot Contorller" (where can I correct this type??) a popup appears saying "Connect failed - Reconnecting"; the reconnecting part nevers seems to succeed.
What am I missing.
I'm using a Pi 3 model B+ and your image with all the software preïnstalled.
Kind regards,
Jean-Pierre