adeept/adeept_picarpro #11: WebServer not autoruning

Nekkit, 2022-09-05:
Hi,

I cant manage to get the webservice to autorun.  It will only work if manually run.
I followed the instructions to install dependencies and even reinstalled os and rerun instructions.

I am running the rasp os from 2021 that was instructed to solve the i2c issue. 

How to check and solve the issue for raspberry pi 4b?

adeept, 2022-09-05:
Hello,
The file of the automatic running program is startup.sh, please check whether the content of the file is correct.
Enter in the Raspberry Pi command line: sudo nano startup.sh
Normally, the contents of this file are:
#!/bin/sh
sudo python3 //home/pi/adeept_picarpro/server/webServer.py

For startup.sh file, see: https://www.adeept.com/learn/tutorial-261.html

Best!

Technical Support
***@***.***

---- Replied Message ----
From ***@***.***>Date 9/5/2022 08:18To ***@***.***>Cc ***@***.***>Subject [adeept/adeept_picarpro] WebServer not autoruning (Issue #11)

Hi,
I cant manage to get the webservice to autorun. It will only work if manually run.
 I followed the instructions to install dependencies and even reinstalled os and rerun instructions.
I am running the rasp os from 2021 that was instructed to solve the i2c issue.
How to check and solve the issue for raspberry pi 4b?

Nekkit, 2022-09-06:
Friend I found the problem.

The setup.py is not programed properly to recognize the relative path of the home folder for a user other than "pi".

in several parts there is a "//home/pi/" route that has implicit that the username is "pi".

By changing the references to "/Home/username/" the instalation created the autorun files correctly.

Now the werServer is runing at start up.

Do note the im using ubuntu and in MozillaFirefox the camera image is frozen, but if I connect to the webserver via Chrome then the video is live.

Ill put here for future reference.