adeept/adeept_picarpro #12: Webserver.py runtimeerror

anggou, 2022-11-09:
I finished assembling and ran it remotely, but it did not open with ip address :5000, so I manually ran webserver.py.

1111@raspberrypi:~/adeept_picarpro/server $
1111@raspberrypi:~/adeept_picarpro/server $ sudo python3 webServer.py
Traceback (most recent call last):
  File "/home/heejoo/adeept_picarpro/server/webServer.py", line 14, in <module>
    import RPIservo
  File "/home/heejoo/adeept_picarpro/server/RPIservo.py", line 17, in <module>
    pwm = Adafruit_PCA9685.PCA9685()
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_PCA9685/PCA9685.py", line 74, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.

However, I got the following error: Please help.

anggou, 2022-11-09:
To solve this problem, I installed the regercy version and also installed the library. After installing the library and restarting the pi, there was a problem that the pi did not turn on again. I'm wondering how to go about solving this problem.

adeept, 2022-11-09:
Hello,
It may be caused by the incompatibility between the Raspberry Pi image file and some dependent libraries.
Please download and burn the Raspberry Pi OS (Legacy) according to the content "Method one" in the link below, then configure SSH and WiFi, and run the setup.py program after downloading the product code.
Link: https://www.adeept.com/learn/tutorial-247.html

Best!

Technical Support
***@***.***

---- Replied Message ----
From ***@***.***>Date 11/9/2022 13:15To ***@***.***>Cc ***@***.***>Subject Re: [adeept/adeept_picarpro] Webserver.py runtimeerror (Issue #12)

To solve this problem, I installed the regercy version and also installed the library. After installing the library and restarting the pi, there was a problem that the pi did not turn on again. I'm wondering how to go about solving this problem.