adeept/adeept_picarpro #13: applying tensorflow

anggou, 2022-11-15:
I want to apply CNN with h5 model to this car camera.
So, I modified the code a bit and tried to use tensorflow.
I did "pip install tensorflow" and ran the code, but an error occurred saying that tensorflow was not installed continuously.
Maybe it's because I installed the legacy version. Could you give me some advice on how to apply tensorflow here?

vsancnaj, 2022-11-22:
Hello @anggou , this is my first time helping here with issues. Could you please direct me were I can find the files or code of what you tried?

LazaUK, 2022-11-30:
Hello @anggou, you may try `pip install --upgrade tensorflow`, so that it upgrades the components installed.

After that, please try these 2 lines of Python code and share the outcomes of your execution:
```
import tensorflow as tf
print(tf.__version__)
```