adeept/adeept_picarpro #14: Issues with setup

MudRockBunny, 2023-06-04:
I am having issues with getting the webserver to function correctly. I have tried all three methods outlined in the tutorial. I have reverted back to method one legacy image as most of the responses i have seen on here end with that being the solution. I am remoted in, have run the setup.py, and attempted to manually run the web server. This the error list i am being given. I am able to cycle the leds via switch.py. No servo response, they do not even center on start up. Here is a list of the errors. What should my next step be?

pi@raspberrypi:~ $ sudo python3 adeept_picarpro/server/webServer.py
Traceback (most recent call last):
  File "adeept_picarpro/server/webServer.py", line 14, in <module>
    import RPIservo
  File "/home/pi/adeept_picarpro/server/RPIservo.py", line 17, in <module>
    pwm = Adafruit_PCA9685.PCA9685()
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 316, in write_byte_data
    self._device.write(data)

MudRockBunny, 2023-06-04:
Also, I am using a RaspberryPi 4

adeept, 2023-06-06:
Hello,
It may be that the HAT board is not installed correctly on the Raspberry Pi, please check whether the HAT and the camera are installed correctly.

Best!

 Technical Support
 ***@***.***
  

---- Replied Message ----
From  ***@***.***> Date  6/5/2023 05:24 To  ***@***.***> Cc  ***@***.***> Subject  [adeept/adeept_picarpro] Issues with setup (Issue #14)  

I am having issues with getting the webserver to function correctly. I have tried all three methods outlined in the tutorial. I have reverted back to method one legacy image as most of the responses i have seen on here end with that being the solution. I am remoted in, have run the setup.py, and attempted to manually run the web server. This the error list i am being given. I am able to cycle the leds via switch.py. No servo response, they do not even center on start up. Here is a list of the errors. What should my next step be?
***@***.***:~ $ sudo python3 adeept_picarpro/server/webServer.py
 Traceback (most recent call last):
 File "adeept_picarpro/server/webServer.py", line 14, in
 import RPIservo
 File "/home/pi/adeept_picarpro/server/RPIservo.py", line 17, in
 pwm = Adafruit_PCA9685.PCA9685()
 File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 75, in init
 self.set_all_pwm(0, 0)
 File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
 self._device.write8(ALL_LED_ON_L, on & 0xFF)
 File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 114, in write8
 self._bus.write_byte_data(self._address, register, value)
 File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 316, in write_byte_data
 self._device.write(data)